os-fpga/FOEDAG #1: MIT license might be a problem

MidsummerNight, 2021-10-05:
Since we're going to use Qt's open source and @tangxifan asked about our GUI's license, I just checked [Qt Open source Licensing FAQ](https://www.qt.io/faq/tag/qt-open-source-licensing) and [Qt Open Source Obligations of the GPL and LGPL](https://www.qt.io/licensing/open-source-lgpl-obligations). MIT licensed software can be integrated with GPL software, though not being a law expert and lacking experience with mixed-license open source projects, I'm not sure if that means we can take GPL/LGPL code from Qt (headers, libraries, modules etc.), modify it and then make it MIT, or do we have to keep our own MIT code strictly seperated from Qt's GPL/LGPL code, or how else should this be properly handled. As the definitive versions of all the above licenses are in English, I recommend someone in the States consult a law expert familar with this field. Keep in mind that GPL/LGPL both have several versions so that's another thing to watch out for. If consulting a lawyer is not viable, maybe we should go with LGPLv3 for the peace of mind?

tangxifan, 2021-10-05:
@MidsummerNight This is indeed to be seriously considered.

I did check the difference between GPL and LGPL. Here is what GNU website says at http://www.gnu.org/licenses/why-not-lgpl.html

```
The choice of license makes a big difference: using the Lesser GPL permits use of the library in proprietary programs; using the ordinary GPL for a library makes it available only for free programs.
```

LGPL looks o.k. to me if we do not modify any Qt source codes under the LGPLv3 license. Otherwise, we have to follow the term: 

```
If you modify a copy of the Library, and, in your modifications, a facility refers to a function or data to be supplied by an Application that uses the facility (other than as an argument passed when the facility is invoked), then you may convey a copy of the modified version:

a) under this License, provided that you make a good faith effort to ensure that, in the event an Application does not supply the function or data, the facility still operates, and performs whatever part of its purpose remains meaningful, or
b) under the GNU GPL, with none of the additional permissions of this License applicable to that copy.
```

My understanding is
- If we use the Qt library as is, and we just link to its compiled library (Think about in Github Actions, we install Qt packages through ``apt-get``), we should be fine to use any license MIT/Apache.
- If we have to modify any Qt source files, we have to put these files in this repo. It is safe to use LGPLv3 license. If any commercial projects use this open-source project, it does not force vendors to open-source the projects.
- **We should avoid to use the GPL codes of Qt sources which will cause the whole project to be under GPL**

Anyway, I will try to ask any expert on this topic.

MidsummerNight, 2021-10-09:
@tangxifan I heard you already bought Qt's commercial license? If we're using commercial Qt then this GUI probably shouldn't be open-sourced at all and we'll have to use a Github private repository or something not public to develop. I've reached out to Qt sales representatives asking for details on this closed-commercial/open-source matter and will update when I get answers.

tangxifan, 2021-10-19:
@MidsummerNight My understanding is that Qt's source codes are under LGPL and GPL. It is free to use them as long as you are doing researches or anything non-commercial.
However, if you want to use it for commercial purpose, you need licenses. Otherwise, you may run into legal issues.
The commercial Qt is mainly on the QtCreator and some GUI support which can ease your UI design, without writing codes

MidsummerNight, 2021-10-19:
@tangxifan Actually GPL regulates your source code, not binaries. For example companies like Redhat do [sell compiled binaries of their Linux products](https://www.redhat.com/en/store/linux-platforms) (of which the source code is GPL-licensed) and that is perfectly legal. The GPL license only requires them to always offer the full source code of their products to the public, and does not care how the binaries created from the source code are used, which is why other RHEL-based distros developed by teams outside of Redhat like Oracle Linux and Alma Linux are able to exist in the first place.

By the way Qt sales responded reponded to me and said that both versions of Qt (commercial license and open source license) can be used for any purposes, be it commercial or not. The differences are:

1. Qt commercial license allows you to choose if you want your project to be open-sourced, whereas the open source license bounds you to the GPL/LGPL obligations.
2. Qt commercial enjoys better official support in terms of bugfixes among other improvements, whereas the open source version would only receive 2 minor updates for every feature update. Take the latest Qt version (6.2) as an example. Commercial Qt 6.2 is an LTS (Long-Term-Support) release that will be supported for many years down the line by Qt, but Open Source Qt 6.2 will only get updated to 6.2.2. This differentation happens at the source code level, unlike Linux distros they will not open source patches later than 6.2.2, so there's no way we can download, compile and apply those patches ourselves. The KDE Project team does [make their own patches for open source Qt](https://invent.kde.org/qt), though they are understandably trailing behind Qt's latest versions.
3. There are certain performance analyzation tools and special modules exclusive to commercial Qt.
4. Both commercial and open source Qt are developed from the same code base. I don't know how they are able to effectively dual-license 

In summary, LGPL/GPL doesn't limit the purposes of our final product. It all comes down to whether we want our own source code to always be fully open and other people forking our code to keep their modifications open as well.

MidsummerNight, 2021-10-20:
@tangxifan That would depend on how well we can isolate this GUI from potential backends developed by people using our GUI. [Android](https://source.android.com/setup/start/licenses), for example, runs on top of Linux kernels (GPL) but most of AOSP's (Android Open Source License) code is Apache 2.0. Smartphone vendors like Samsung, Xiaomi take the AOSP code, add their own customizations (resulting in OneUI and MIUI skins), then let it run on kernels provided by SoC vendors (though phone vendors have to add their own tweaks to the vendor kernels too). In the end, phone vendors only need to release the source code of the kernels, but can keep the rest of the software (UI skins) to themselves. It's true that GPL can make things less flexible for corporate members though, since they would have to seperate closed-source code from our GPL GUI very carefully.

I just came across this company called [FOSSA](https://fossa.com/product/open-source-license-compliance) that seems to offer open source license compliance analysis services, not sure if they can be of help.