os-fpga/FOEDAG #68: Add "make doc" target

alain-rs, 2021-11-01:
**Is your feature request related to a problem? Please describe.**
I already forgot the steps to compile the docs, a make doc target is needed in the top-level Makefile

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

alain-rs, 2021-11-01:
I also need help to format the doc, with the markup language

alain-rs, 2021-11-01:
I'm seeing this, are we missing dependancies for redthedocs? If so they need to be in the install guide too.

make doc
cd docs && make html
make[1]: Entering directory '/home/alain/FOEDAG/docs'
Running Sphinx v3.5.4

Extension error:
Could not import extension m2r2 (exception: No module named 'm2r2')
make[1]: *** [Makefile:30: html] Error 2
make[1]: Leaving directory '/home/alain/FOEDAG/docs'
make: *** [Makefile:132: doc] Error 2
alain@alain-xps:~/FOEDAG$ m2r2
Command 'm2r2' not found, did you mean:
  command 'm2r' from deb m2r (0.2.1-5)
Try: sudo apt install <deb name>

tangxifan, 2021-11-01:
Yes. We should document it.

If you go to the ``docs`` directory, and run

```
pip3 install -r requirements.txt
```

The dependencies will be installed.

My question is 
- Should it in the Makefile or only in the documentation?

alain-rs, 2021-11-01:
It should be with the INSTALL instructions.

OK, it works now but there are warnings:
make doc 
cd docs && make html
make[1]: Entering directory '/home/alain/FOEDAG/docs'
Running Sphinx v3.5.4
making output directory... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 11 source files that are out of date
updating environment: [new config] 11 added, 0 changed, 0 removed
checking for /home/alain/FOEDAG/docs/source/z_reference.bib in bibtex cache... not found                                                                 
parsing bibtex file /home/alain/FOEDAG/docs/source/z_reference.bib... parsed 1 entries
reading sources... [100%] tutorials/index                                                                                                                
/home/alain/FOEDAG/docs/source/contact.rst:4: WARNING: duplicate label contact, other instance in /home/alain/FOEDAG/docs/source/contact.rst
source/help/help.txt:14: WARNING: Definition list ends without a blank line; unexpected unindent.
/home/alain/FOEDAG/docs/source/tutorials/getting_started/dependencies.rst:45: WARNING: Duplicate explicit target name: "install_ubuntu_dependencies_build".
/home/alain/FOEDAG/docs/source/tutorials/getting_started/dependencies.rst:26: WARNING: Duplicate target name, cannot be used as a unique reference: "install_ubuntu_dependencies_build".
/home/alain/FOEDAG/docs/source/tutorials/getting_started/dependencies.rst:40: WARNING: Unknown target name: "install_macos_dependencies_build".
looking for now-outdated files... none found
pickling environment... done
checking consistency... /home/alain/FOEDAG/docs/source/reference.rst: WARNING: Citation [XTang_FPL_2019] is not referenced.
done
preparing documents... done
writing output... [100%] tutorials/index                                                                                                                 
generating indices... genindex done
writing additional pages... search done
copying images... [100%] overview/figures/gui_top_architecture.png                                                                                       
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 6 warnings.

The HTML pages are in build/html.
make[1]: Leaving directory '/home/alain/FOEDAG/docs'
cd -
/home/alain/FOEDAG/docs