os-fpga/FOEDAG #74: Add a step to generate the installation package

liibin, 2021-11-02:
**Is your feature request related to a problem? Please describe.**
 I think we should add a step to generate the installation package

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tangxifan, 2021-11-02:
@liibin Can you explain more details about what the installation packages are? Currently, in the INSTALL.md, if you run make install, you should have binaries installed. This is also in the documentation.

https://foedag.readthedocs.io/en/latest/tutorials/getting_started/build_instructions/#compile-source-codes

liibin, 2021-11-03:
It can be installed and run directly without compiling environment。

tangxifan, 2021-11-03:
> It can be installed and run directly without compiling environment。

OK. What you mean is package installer. It is not in the domain of this open-source project. You can find some options from Linux and Windows communities. In Linux, there is a free one while in Windows, you may have to purchase one.

alain-rs, 2021-11-03:
I think we can build artefacts in github that contain pre-built executables.
But this can be done in the internal repos that contain device specific data that need to be shipped,
not on the open-source repo that does not contain all that needs to be shipped to make a release.

Alain Dargelas, Ph.D.
Vice President EDA Software
RapidSilicon

OK. What you mean is a package installer. It is not in the domain of this open-source project. You can find some options from Linux and Windows communities. In Linux, there is a free one while in Windows, you may have to purchase one.

closed 2022-08-26 by ravic-rs