os-fpga/FOEDAG #77: About create new projects

liibin, 2021-11-03:
I intend to implement the following functions on the dialog.

New project wizard
    Project Location
        project name(default project_1)
        project location (support create subdirectory)
        
    Project Type
        RTL project (default)
        Post-synthesis project
        
    Add Design Source Files
        Add files
        Add directories
        Create file
            file name
            file type
            file path
        Display source file list
        Remove
        Move up
        Move Down
        Support copy design files to project
        
    Add Constraints
        Add files
        Create file
            file name
            file type
            file path
        Display source file list
        Remove
        Move up
        Move Down
        Support copy files to project
        
    Select device/board
        Display the device/board list(Read data form a device/board file(like *.xml))
        Support search by family, device，package and so on.
        
    Summary
        Displays the details of the project you are about to create.
    
The project type and file type can be configured through a configuration file(like *.xml).

[image]

After clicking finish, the project file and project directory will be generated.As follows:
[image]
[image]

About the widget window floorplanning ,I will do it soon by QT designer.

Open to discussion here.

alain-rs, 2021-11-03:
- An auto-population of the RTL design list is a must-have feature.
Please have a way to select directories (Multi-selection) and automatically add all the files with extensions (.v .sv .vhd - can be configured) to the project. 
- Compilation directives need to be captured in the project wizard (make YOSYS=1) and passed to the Synthesis tool.

tangxifan, 2021-11-03:
@alain-rs @liibin @MidsummerNight @KKaihui 

This is a good start. Since we will have more widgets to be merged, it is important to have a template for developers to follow when proposing a new widget.

Here is what I propose.
The template will consist of the following sections

## Motivation
A short statement why we need this widget 

## Technical Details
Detail the technicality of this widget, it must include the following aspects.
### Code flow
Prefer to use a flow chart to show how the widget is going to be integrated into the framework, especially
- What are the input files or data structures that it will interface
- What are the output files or data structures that it will interface
- What are the critical steps inside the flow to accomplish the target

### Tcl command
What is the tcl command to be created or updated?
- The tcl command is the equivalent to the widget functionality. It will be used to replay the GUI in regression tests
- What are the options will be available in the tcl command
- What are the dependent tcl commands? (e.g., it must only be called after which tcl commands)

### File formats
If you do have input and output files, please specify the syntax in each file format.

## Technical plan
You need to state the estimated timeline to accomplish the target, especially 
- What are the pull requests you will create (you do not need to be very accurate here but you need to state roughly what these pull requests will do, what is the dependencies on the pull requests). Please remember to avoid large pull requests
- What will be your updates on documentation
- What will be your updates on regression tests

**Let me know what you think. Once agreed, I will add these to our documentation as our developer guidelines. I will also update our feature request templates to guide developers.**

liibin, 2021-11-04:
@tangxifan I remember you said each widget is an executable binary，so，how the widget is going to be integrated into the framework. Can you give me an example.

tangxifan, 2021-11-04:
> @tangxifan I remember you said each widget is an executable binary，so，how the widget is going to be integrated into the framework. Can you give me an example.

You can just create a separated cpp file where you instanciate the widget object. See Alain's example:

https://github.com/os-fpga/FOEDAG/blob/main/src/Tcl/HelloTcl_test.cpp

It is typically used for unit tests.

alain-rs, 2021-11-04:
Correct, the idea is that the Widget should be able to be instantiated in a simple main.cpp or unit test, and compile with all symbols resolved using a minimal set of other subsystems, so it can be tested individually.

liibin, 2021-11-05:
@tangxifan @alain-rs Thank you very much,i will have a try.

liibin, 2021-11-10:
> @alain-rs @liibin @MidsummerNight @KKaihui
>
> This is a good start. Since we will have more widgets to be merged, it is important to have a template for developers to follow when proposing a new [...]

liibin, 2021-11-10:
We had a discussion with @tangxifan 
## Motivation
Provides a wizard for users to create projects

## Technical Details
Detail the technicality of this widget, it must include the following aspects.
### Code flow

[image]

### Tcl command
The new project wizard is a widget in the GUI. To validate the correctness of GUI, unit tests is required. In the TCL command, we only need a command which stores the results of the widget. For example, the new project wizard should output a project file after user clicking the ‘finish’ button.
So running a new project wizard is equivalent to run the following commands:

> read_device_list --file devices.xml (This command should be loaded when initialize the engine)
> create_project  --file project.xml

### File formats
**project.xml format**
```
<project type=”[rtl | post-implementation | …]” name=”<string>”>
	<device name=” x7c100t”/>
	<sources>
		<source id=”0” type=”[verilog | vhdl | systemverilog]” file=”~/counter.v” is_top=”[true|false]”/>
         … more <source>
	</sources>
	<constraints>
		<constraint id=”0” type=”sdc” file=”~/counter.sdc”/>
		<constraint id=”1” type=”odc” file=”~/counter.odc”/>
         … more <constraint>
	<constraints>
	<testbenches>
		<testbench id=”0” type=”[verilog | vhdl | systemverilog]” file=”~/counter_tb.v”/>
		… more <testbench>
	</testbenches>
</project>
```

**device.xml format**
```
<device_list>
    <device name=”x7c100t” series=”7series” family=”artix7” package=”SBG484” pin_count=”238” speedgrade=”1” core voltage=”1.1V”>
        <resource type=”io” num=”200”/>
        <resource type=”lut” num=”8000”/>
        <resource type=”ff” num=”16000”/>
        <resource type=”bram” num=”48”/>
    </device>
    <!-- more devices …  -->
</device_list>
```

## Technical plan

Pull request | content |  
-- | -- | --
First | Add the  project GUI | finished
Second | device list and project data structure |  
Third |  Create folders, project file |  
Fourth | TCL command |

alain-rs, 2021-11-10:
There is another set of Tcl commands that are necessary to perform GUI automated testing like the already existing gui_start gui_close commands:
```
set wid [gui_project_create open]                          << Returns a pointer to the Widget created
gui_set_field $wid "device_name" <part_name>   << Set a field with a value inside the Widget
gui_click $wid "Next"      << "Virtually" click the Next button in the form
...
gui_project_create close $wid
```

liibin, 2021-11-10:
> There is another set of Tcl commands that are necessary to perform GUI automated testing like the already existing gui_start gui_close commands:
>
> ```
> set wid [gui_project_create [...]

Like this?
[image]

alaindargelas, 2021-11-10:
Yes! Excellent.

alain-rs, 2021-11-10:
Yes, Excellent!
You can create GUI testing scripts like:
https://github.com/os-fpga/FOEDAG/blob/main/tests/TestGui/gui_start_stop.tcl

And run them part of the automated GUI test: 
https://github.com/os-fpga/FOEDAG/blob/df8bc7e42f782336b1a0826f4119206a5ce5c9ab/Makefile#L122

closed 2021-11-19 by liibin