os-fpga/FOEDAG #80: Main branch msvc starts failing with no code change

alaindargelas, 2021-11-04:
> **Describe the bug**
> A clear and concise description of what the bug is.

> **To Reproduce**
Simply rerun one of the runs from main that used to pass, and observe the msvc failure, something about a .vc file syntax and multiple processes accessing a file. 
Looks like something changed in the compilation environment.

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

alain-rs, 2021-11-04:
@liibin do you see this problem when you compile locally on Windows? Or is it just a github issue?

tangxifan, 2021-11-04:
@alain-rs I experienced some changes on Github runner machines without any notification from Github. This time it may be the same. Last time, I just wait for 2 weeks, the problem was resolved (Github did something under table).

liibin, 2021-11-05:
> @liibin do you see this problem when you compile locally on Windows? Or is it just a github issue?

I encountered an out of memory error. After I close other apps, the compilation is successful and can run.

alaindargelas, 2021-11-05:
Thanks, so it is a change in github environment.
We are going to set our own runner actions for Windows on our servers as soon as possible.

alain-rs, 2021-11-05:
The latest version of cmake used by github has a bug, 
if I fix the version of cmake to the previous version, msvc runner works again....

alain-rs, 2021-11-05:
Fixed by #79

closed 2021-11-05 by alain-rs