os-fpga/FOEDAG #88: Warning cleaup

alain-rs, 2021-11-12:
> **Describe the bug**
> A clear and concise description of what the bug is.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone repository and checkout commit id: <The problem commit id>
> 2. Execute a test case or your own example: <more details>
> 3. See error

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

Please cleanup all warnings so we can turn on -Werror compiler flag and treat all warnings as errors.

alain-rs, 2021-11-12:
In Main and Newproject subsystems

alain-rs, 2021-11-12:
I temporarily added the following to mute the warnings, but they need to be fixed:
https://github.com/os-fpga/FOEDAG/blob/83fd84e0f1fb529451ba4b9830247f91377f37ff/src/NewProject/CMakeLists.txt#L11

alaindargelas, 2021-11-15:
Done

alain-rs, 2021-11-15:
Done

closed 2021-11-15 by alain-rs