os-fpga/FOEDAG #97: About project navigator

liibin, 2021-11-18:
## Motivation
Displays the hierarchy of files contained in the project.
Provide GUI for project management.

## Technical Details
Detail the technicality of this widget, it must include the following aspects.
### Code flow
[image]

### Tcl command
We can also manage the project by using Tcl command.

```
> create_fileset [<type>][<setname>]

> add_files [<type>][<filename>] 

> set_active [<type>] [<setname>]

> set_as_top [<type>][<filename>]

> set_as_target [<filename>]
```

The tcl command operates on the active fileset.

About the widget window floorplanning ,I will do it soon by QT designer.

Open to discussion here.

alain-rs, 2021-11-18:
If possible we need to be able to support multiple designs in one Gui session. It will come handy to perform things like what-if analysis or seed sweeping.

So a create_design <name> function is necessary, then every other function can operate on the last design created.
For switching among designs a set_current_design <name> function will  be necessary too.

alain-rs, 2021-11-18:
It looks like your fileset is the design concept, then let's do create_design instead.
What is <type> for ? SV, VHDL?

create_design <designName>
set_active_design <designName>
set_top_module <moduleName>
add_files <filename>....

What is set_as_target?

liibin, 2021-11-19:
set_as_target :  set target constraint file.

I think we should take another look at the project structure.
```
<Project Path="/root/Desktop/project_1/project_1.ospr">
    <Configuration>
        <Option Name="ID" Val="20211118173933936"/>
        <Option Name="ActiveSimSet" Val="sim_1"/>
        <Option Name="Project Type" Val="RTL"/>
    </Configuration>
    <FileSets>
        <FileSet Name="constrs_1" Type="Constrs" RelSrcDir="/project_1.srcs/constrs_1">
            <File Path="$OSRCDIR/project_1.srcs/constrs_1/constr1.SDC"/>
            <File Path="$OSRCDIR/project_1.srcs/constrs_1/constr2.SDC"/>
            ... more <File/>
            <Config>
                <Option Name="TargetConstrsFile" Val="$OSRCDIR/project_1.srcs/constrs_1/constr2.SDC"/>
            </Config>
        </FileSet>
        <FileSet Name="sim_1" Type="SimulationSrcs" RelSrcDir="/project_1.srcs/sim_1">
            <File Path="$OSRCDIR/project_1.srcs/sim_1/simsrc1.v"/>
            <File Path="$OSRCDIR/project_1.srcs/sim_1/simsrc2.v"/>
            ... more <File/>
            <Config>
                <Option Name="TopModule" Val="$OSRCDIR/project_1.srcs/sim_1/simsrc1.v"/>
            </Config>
        </FileSet>
        <FileSet Name="sources_1" Type="DesignSrcs" RelSrcDir="/project_1.srcs/sources_1">
            <File Path="$OSRCDIR/project_1.srcs/sources_1/dsrc1.v"/>
            <File Path="$OSRCDIR/project_1.srcs/sources_1/dsrc2.v"/>
            ... more <File/>
            <Config>
                <Option Name="TopModule" Val="$OSRCDIR/project_1.srcs/sources_1/dsrc2.v"/>
            </Config>
        </FileSet>
        ... more <FileSet/>
    </FileSets>
    <Runs>
        <Run Name="imple_1" Type="Implementation" SrcSet="sources_1" ConstrsSet="constrs_1" State="current" SynthRun="synth_1"/>
        <Run Name="synth_1" Type="Synthesis" SrcSet="sources_1" ConstrsSet="constrs_1" State="current" SynthRun="">
            <Option Name="Compilation Flow" Val="Classic Flow"/>
            <Option Name="Device" Val="x7c100t"/>
            <Option Name="Family" Val="artix7"/>
            <Option Name="LanguageVersion" Val="SYSTEMVERILOG_2005"/>
            <Option Name="Package" Val="SBG484"/>
            <Option Name="Series" Val="7series"/>
            <Option Name="TargetLanguage" Val="VERILOG"/>
        </Run>
        ... more <Run/>
    </Runs>
</Project>
```
On the Run node, two FileSets (srcset = "sources_1" constrsset = "constrs_1") can be found through two values of type = "synthesis" and state = "current".

Each fileset has multiple files. You can find the TopModule and TargetConstrsFile on the Config node.

alain-rs, 2021-11-19:
OK got it.

closed 2021-11-29 by liibin