os-fpga/FOEDAG #105: Text Editor

liibin, 2021-11-30:
## Motivation
Provides an interface for editing HDL code.

## Technical Details

### Code flow
[image]

### Basic functions

- HDL syntax highlight and error indicator
- Multi-tab to open multiple files
- Basic editing operations,such as cut,copy,paste,delete,undo,redo.

I will use a third-party open source library: QScintilla。
https://www.riverbankcomputing.com/static/Docs/QScintilla/

Open to discussion here.

alaindargelas, 2021-11-30:
Must have feature: 
  - Search across entire project RTL source code
     - Find all module instances of a module definition
     - find all occurrence of a text 
  - Link to/from hierarchy tree viewer
Nice to have features: Auto-completion (Might require language server)

liibin, 2021-11-30:
> Must have feature:
> 
> * Search across entire project RTL source code
>   
>   * Find all module instances of a module definition
>   * find all occurrence of a text
> * Link to/from hierarchy tree viewer
> 
> Nice to have features: Auto-completion (Might require language server)

OK，I'll add this feature.

closed 2021-12-08 by liibin