os-fpga/FOEDAG #121: Console

KochynVolodymyr, 2021-12-21:
Motivation.
-----------
Provide tcl console for the GUI.

Plan
---------
First step: 
    Standalone application with full featured tcl console. The console should behave like regular console. 
Second step: 
    Integration into main GUI. 

Requirements for standalone: 
------
- When start gui – no need tcl interpreter anymore. 
- Tcl console should be a part of the project as well as standalone 
- Main functionality:
-- Text field for the input and output like standard console
-- The widget doesn’t allow to enter command while running smth. 
-- There is should be possibility to terminate current run. 
- Common console functionality: 
-- Abort current running. 
-- History  
-- Possibility to run prev command with !<cmd num> 
-- Autocompletion by tab
-- Autocomplete for command parameters is more like nice to have feature.
-- In case multiple choice show all cases 
-- Clear 
- Search 
- Hyperlinks (e.g., file line) 
- The command should run in some thread in order not block gui. 
- Multiline command like regular console – more like nice to have feature 
- Interface between console and main part: 
TBD 

Libraries for command line functionality: 

    Readline 

    https://github.com/JuliaAttic/readline port for windows 

    Libedit  

    Libeditline is fork for C++. 

    Linenoise  

    linenoise-ng C++ with window support but the repo is archiched 

UML diagram:

[image]

**ConsoleWidget**. Main widget.
**ConsoleInterface**. Abstract class that provide interface for the widget.
**TclConsole**. Inmplementation of the ConsoleInterface for the TCL.
**TclWorker**. Class that encapsulate tcl. Must run in separate thread.
 
Requirements for main application: 
--------
- Possibility to add and remove console. 
- TBD

 

 

Open questions: 
    What is A filter to display only errors and warnings

alain-rs, 2022-01-12:
@KochynVolodymyr, Please add a gui test that exercises the console features, including evaluating tcl commands like:

> set FOO 1
> batch { incr FOO }
> update_result
> puts $FOO    << Should be 2

Or evaluating an incorrect script and making sure the correct error shows:
> bad_command
invalid command name "bad_command"

ravic-rs, 2022-09-01:
This is complete, closing.

closed 2022-09-01 by ravic-rs