os-fpga/FOEDAG #141: Console: autocomplete for the command arguments

KochynVolodymyr, 2022-01-25:
User wants to have autocomplete for arguments of the command like switches, filenames, dirs or keywords. As an example could be 'git' in the bash.

Currently this issue isn't urgent.

KochynVolodymyr, 2022-06-27:
@alain-rs TCL itself doesn't have any help for commands arguments. In order to implement this we have to:
* write some subsystem for command arguments processing
* fill some map with all arguments  for every command.

This will take a while and requires further support. 
vivado doesn't have this feature.
Do we need it now?

alaindargelas, 2022-06-27:
I think the most important commands that need completion are actually the commands like synth_opts, synth.

Tcl built-in commands or user defined proc arguments can be discovered using the Tcl introspection (info command ....) you can retrieve their arguments and create an argument map db per command at start time as you mentioned, using a tcl script that gets evaluated once. There are examples of such scripts that get invoked by the C++ code and those output gets read back in C++ already in the TclInterpreter.

For the C++ registered commands like synth_opts, synth... we need also to build that mechanism during the registration of the command in the register commands functions.

This is a nice usability feature. Not urgent.

KochynVolodymyr, 2022-06-27:
I didn't find a way how to retrieve arguments, can you help? info commands... gives only the command name, but not the arguments

KochynVolodymyr, 2022-06-27:
That works for procedures only, how about commands?