os-fpga/FOEDAG #153: Console eats carriage returns

alain-rs, 2022-01-31:
> **Describe the bug**
> A clear and concise description of what the bug is.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone repository and checkout commit id: <The problem commit id>
> 2. Execute a test case or your own example: <more details>
> 3. See error

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

The console eats carriage returns
This command works:

batch { synth ; globp }

That does not do the second command:

batch {
synth
globp
}
The carriage return in between synth and globp is gone when it reaches here:
https://github.com/alain-rs/FOEDAG/blob/b9d16b1d0b5effe5dfc910fd8fbf85777c75e4b5/src/Compiler/Compiler.cpp#L182

KochynVolodymyr, 2022-02-01:
This is standard console behavior. If you try in the standard console run something like this:

echo hello \
echo world 

result command: echo hello echo world
output: hello echo world

so the user must put ';':
batch {
synth;
globp
}

What do you think?

alaindargelas, 2022-02-01:
It is not standard Tcl console behavior (Which matters here):

tclsh
% proc myproc { } {
   puts HERE
}
% myproc
HERE
%

KochynVolodymyr, 2022-02-01:
Do you want to keep it multiline in the history?

alaindargelas, 2022-02-01:
Yes, else it would not be able to re-execute correctly.

alain-rs, 2022-02-07:
This seems to be fixed.

KochynVolodymyr, 2022-02-07:
Please reopen, I'm working on it. It's not ready jet.

PR #177 (2022-02-14): Issue 153 multiline improve
Close #153
commit f87fc3d9 (2022-02-14): Keep multiline command as multiline in the history. Close #153
closed 2022-02-14 by alain-rs