os-fpga/FOEDAG #166: CentOS CI support

KochynVolodymyr, 2022-02-08:
Add action(s) to run FOEDAG on the CI within the CentOS.

tangxifan, 2022-02-08:
You may refer to this workflow https://github.com/OPM/ResInsight/blob/dev/.github/workflows/centos7.yml

alain-rs, 2022-02-14:
I reached an impasse as CentOS7 is the only Centos supported by Github. There is no CentOS8 support.
On Centos7, the built-in package for QT5 is older (5.9) and QScintilla requires QT5.11
/__w/FOEDAG/FOEDAG/third_party/QScintilla-2.13.1/src/Qsci/qsciglobal.h:33:2: error: #error "Qt v5.11.0 or later is required"
   33 | #error "Qt v5.11.0 or later is required" 
https://github.com/alain-rs/FOEDAG/runs/5172152642?check_suite_focus=true
I tried to build a later QT in the CI, but failed with weird gzip errors.

KochynVolodymyr, 2022-02-15:
I was able to solve this problem but got error during build qt.
https://github.com/KochynVolodymyr/FOEDAG/runs/5201147694?check_suite_focus=true

alain-rs, 2022-02-22:
Fixed

closed 2022-02-22 by alain-rs

KochynVolodymyr, 2022-02-22:
Do you think wait 2 hours to finish job is ok? I don't like this idea. Let's discuss how we can make it shorter

alain-rs, 2022-02-22:
Opened issue #191 to track this.
At any rate, once we have the OpenFPGA backend attached to the GUI, the CI is going to become drastically longer (hours) for all platforms. But agreed, Centos should not be another 2 hours longer than the Ubuntu CI which is going to be probably 2 hours at that time.