os-fpga/FOEDAG #199: Clicking in the console while command running: bogus error

alain-rs, 2022-02-24:
Clicking in the console while command running: bogus error:
invalid command name "Synthesizing"

# 
# synth
# Synthesizing design: Some cool design...
 0% File: [device.xml](https://github.com/home/alain/FOEDAG/device.xml) just for test
10% File: [.readthedocs.yml](https://github.com/home/alain/FOEDAG/.readthedocs.yml) just for test
20% File: [src](https://github.com/home/alain/FOEDAG/src) just for test
30% File: [.gitignore](https://github.com/home/alain/FOEDAG/.gitignore) just for test
40% File: [Config.cmake.in](https://github.com/home/alain/FOEDAG/Config.cmake.in) just for test
50% File: [README.md](https://github.com/home/alain/FOEDAG/README.md) just for test
60% File: [cmd.log](https://github.com/home/alain/FOEDAG/cmd.log) just for test
70% File: [.git](https://github.com/home/alain/FOEDAG/.git) just for test
80% File: [.gitmodules](https://github.com/home/alain/FOEDAG/.gitmodules) just for test
90% File: [tests](https://github.com/home/alain/FOEDAG/tests) just for test
Design Some cool design is synthesized!

invalid command name "Synthesizing"
# 
#

KochynVolodymyr, 2022-02-25:
closed by #202

KochynVolodymyr, 2022-02-25:
@alain-rs please check and close

closed 2022-02-25 by alain-rs