os-fpga/FOEDAG #218: New project action

KochynVolodymyr, 2022-03-04:
Currently we can start new project only once.
Possible solutions:
* Open another instance of the whole FOEDAG project. Create another process.
* Open another main widget within same process.

alain-rs, 2022-03-04:
The tool should only support one project at a time. 
If a new project is needed, prompt the user to close the previous project, then open the new project.
Within one project, we can support multiple compilation runs with different compilation options, but only one project.

ravic-rs, 2022-09-01:
@KochynVolodymyr I believe we have a solution for this here https://github.com/os-fpga/FOEDAG/pull/549
Please confirm.

KochynVolodymyr, 2022-09-01:
> @KochynVolodymyr I believe we have a solution for this here #549 Please confirm.

No. We have another solution - Close project. But it is kind of weird behavior and out of user experience.