os-fpga/FOEDAG #237: New project window features

alain-rs, 2022-03-09:
- Be able to change the order of source file
- Be able to remove source file
- From the Project Viewer, be able to re-open the project wizard of an existing project and edit any fields from the wizard.
- Create the project in the local directory from where the tool was launched, or give a directory finder choice, not in $HOME

liibin, 2022-03-10:
> * Be able to change the order of source file
> * Be able to remove source file

It has been realized through these three buttons.   I'll change the word delete to remove.
[image]

alain-rs, 2022-03-10:
Thanks!
This feature is still missing:
From the Project Viewer, be able to re-open the project wizard of an existing project and edit any fields from the wizard.

When clicking in the Project Viewer instead of having the "Add files" open just the Wizard at the "add file" view, open the entire wizard and allow to click "Finish" on the current view.

In the Project Viewer, the "create design" menu does not seem to work, the "design" panel should open to show the created design.

In the Project Viewer, there is no "Add Constraints" menu, it points to the "Add files" menu, open the wizard to the constraint view in the wizard.

alain-rs, 2022-03-11:
@liibin, let me know if you are going to work on these items, we need to get the project definition working well very soon.
Please try different scenarios (Testing).

liibin, 2022-03-15:
> @liibin, let me know if you are going to work on these items, we need to get the project definition working well very soon. Please try different scenarios (Testing).

I'm busy with the company's annual reports.I am going to work on these items next week.

ravic-rs, 2022-08-17:
Closing this as the feature update branch has been merged.

closed 2022-08-17 by ravic-rs