os-fpga/FOEDAG #260: Hook Project db with Design object

alain-rs, 2022-03-29:
@liibin, please provide comments.
@volodymyrkochyn you should implement

@liibin has developed the New Project wizard, it creates some internal representation of the de sign (File list).
It does not support corresponding Tcl commands.
In the Compiler subsystem, there is a Design object that should model all the necessary components of a design (Files, top level module name, constraint file).  It also register the official Tcl commands to create a design:
create_design
set_top_module
add_design_file 
add_constraint_file (to be added).

We need to sync the GUI object model from the forms to the Design object.

The tcl commands like add_design_file should sync with the GUI forms too.

KochynVolodymyr, 2022-04-01:
Hi! I see what is the issue and have already started work. Unfortunately there is no InProgress status for issues.

KochynVolodymyr, 2022-04-03:
Added two new commands:
* set_as_target
* create_file_set

ravic-rs, 2022-09-01:
Closing as done.

closed 2022-10-19 by alain-rs