os-fpga/FOEDAG #262: Stop button needs to stop any task

alain-rs, 2022-03-30:
The new CompilerOpenFPGA::Synthesize method runs yosys as a separate process.
Need to kill the process or close the pipe (Which should kill the process) when the stop command is pressed.

alain-rs, 2022-03-31:
Run 
./build/bin/raptor --compiler openfpga --script tests/Testcases/raygentop/raygentop.tcl

Stop the process using the stop button. 
There is an error message from QT

closed 2022-03-31 by alain-rs

KochynVolodymyr, 2022-03-31:
@alain-rs is this issue solved?

alain-rs, 2022-04-01:
Yes.

nadeemyaseen-rs, 2022-04-29:
@alain-rs  I have tested it today with `FOEDAG/tests/Testcases/aes_decrypt_fpga/aes_decrypt.tcl`, the process did not stop on pressing the stop button in raptor.

KochynVolodymyr, 2022-04-29:
Stop button never work in case --script.

alaindargelas, 2022-04-29:
@KochynVolodymyr the stop button should work for all the commands "synth, pack, place, route..." regardless if they where typed manually or running through a script.