os-fpga/FOEDAG #268: Console issues

alain-rs, 2022-04-04:
When running a large trace, the console systematically has error message on the 1rst carriage return,
Install latest opensource yosys, and do "make test/openfpga" 

Also, if mouse is pointing to text above, has a hard time to come back to focus on the last line (the prompt line).
Arrow up also is out of screen (below visible section) when showing history.

KochynVolodymyr, 2022-04-04:
Please give more details for second and third issues. I think it's something existing from the beginning. The first issue is here: https://github.com/os-fpga/FOEDAG/pull/269

alain-rs, 2022-04-05:
1) In the console, If you point the mouse to a text up (scroll), I have a hard time to come back to focus on the last line (the prompt line).

2) If you create a multiline entry in the console (after a long run),
proc mine { } { 
puts HERE
}

Then Arrow up,  the display will be out of screen (below visible section) when showing history.

alain-rs, 2022-04-28:
Looks like it is all fixed

closed 2022-04-28 by alain-rs

KochynVolodymyr, 2022-04-28:
The second item wasn't fixed.