os-fpga/FOEDAG #286: Project not in sync with Tcl

alain-rs, 2022-04-13:
Run:
 
./dbuild/bin/foedag --compiler openfpga --script tests/Testcases/raygentop/raygentop_gui.tcl

I don't understand the result below: where is the source file showing? Why there is a repeat of the project name under the design files?

[image]

KochynVolodymyr, 2022-04-13:
You have approved this. I asked in the mail with screenshots.

alaindargelas, 2022-04-13:
I didn't see the screenshots. Now I see it :-) 
Instead of RAYGENTOP(0) (Active), I should see the actual file 
tests/Testcases/raygentop/raygentop.v with a file folder structure

alaindargelas, 2022-04-13:
There is no need for multiple designs under the Design source.
The only design here is RAYGENTOP.
On the other hand we need to see which source files are under the Design Source section.

The Tcl script in the example loads one source file, we should see (All folded intially):
Design Sources(1)
   {FolderIcon} tests
         {FolderIcon} TestGui
             ....
                {FileIcon}  name of the verilog file    << If right click I should be able to open the file in the source editor

KochynVolodymyr, 2022-04-14:
Project Navigator rework is required here.

KochynVolodymyr, 2022-06-23:
This need to close.

closed 2022-06-23 by alain-rs