os-fpga/FOEDAG #297: Tcl error messages are silent

alain-rs, 2022-04-15:
Edit  tests/Testcases/raygentop/raygentop.tcl, add an illegal command at the beginning of the file,
no error show, simply the prompt
This has to do with the latest project sync, that was working before.
I could show error in a command using the ErrorMessage method, but if the error is an illegal script the Tcl error messaging has been suppresed

./build/bin/foedag --script  tests/Testcases/raygentop/raygentop.tcl --batch

KochynVolodymyr, 2022-04-15:
I tried version month old and before my changes. Looks like it haven't work at all. It doesn't related to sync.
If you have version that works please put here.

KochynVolodymyr, 2022-04-15:
Caused by https://github.com/os-fpga/FOEDAG/pull/293

PR #301 (2022-04-15): Fixed silent error message when run foedag with --script flag (Close …
close #297
commit b16d7162 (2022-04-15): Fixed silent error message when run foedag with --script flag (Close #297) (#301)
closed 2022-04-15 by alain-rs