os-fpga/FOEDAG #299: All error messages originating from a Tcl script (--script <file>) need to show the Tcl error stack

alain-rs, 2022-04-15:
No error stack is showing currently, again some masking issue.
Users have to type puts $errorInfo to see the errors...

KochynVolodymyr, 2022-04-15:
How I can reproduce this. Please put any command. When this works fine.

alain-rs, 2022-04-15:
Create a file:
```
proc a { } {
  b
}
proc b { } {
 not_defined
}
a
```

tchsh shows the full stack trace:

~/FOEDAG$ tclsh t.tcl
invalid command name "not_defined"
    while executing
"not_defined"
    (procedure "b" line 2)
    invoked from within
"b"
    (procedure "a" line 2)
    invoked from within
"a"
    (file "t.tcl" line 7)

FOEDAG only shows the final error. It has to do with $errorInfo not being displayed.

~/FOEDAG$ ./dbuild/bin/foedag --compiler openfpga --script t.tcl --batch
Tcl only mode
invalid command name "not_defined"
%

KochynVolodymyr, 2022-04-15:
Got it! I saw this only in replay mode, we force to output this.  But not in other cases.

alaindargelas, 2022-04-15:
Note that in GUI mode, we want to recognize (file "t.tcl" line 7) as a file location and create an hyperlink to the tcl file

KochynVolodymyr, 2022-04-18:
[image]

[image]

PR #303 (2022-04-25): Add tcl stack trace (close #299)
Close #299
closed 2022-04-25 by alain-rs