os-fpga/FOEDAG #302: stderr shows in the terminal while GUI mode

alain-rs, 2022-04-15:
It should show in the GUI console instead (in red).

alain@alain-xps:~/FOEDAG$ ./dbuild/bin/foedag --compiler openfpga --script tests/Testcases/raygentop/raygentop.tcl 
% Error 1: 
Type: Routing
File: /home/alain/RapidSilicon/Raptor/OpenFPGA_RS/vpr/src/base/vpr_api.cpp
Line: 801
Message: Fixed channel width must be specified when loading routing (was -1)
Error 1: 
Type: Routing
File: /home/alain/RapidSilicon/Raptor/OpenFPGA_RS/vpr/src/base/vpr_api.cpp
Line: 801
Message: Fixed channel width must be specified when loading routing (was -1)
done!
QObject::~QObject: Timers cannot be stopped from another thread
Segmentation fault (core dumped)

KochynVolodymyr, 2022-04-15:
It works for me

alaindargelas, 2022-04-15:
Yes, but the errors show in the terminal.
Have you setup the whole compiler flow? Yosys + VPR? You need to do that to be able to test things properly from there on.

alaindargelas, 2022-04-15:
The messages above are error messages from VPR.
If you don't have VPR you cannot test properly.

KochynVolodymyr, 2022-04-15:
got it! I didn't setup VPR

KochynVolodymyr, 2022-04-18:
[image]

PR #308 (2022-04-26): Redirrect stderr from compiler into the tcl console (close #302)
Close #302
commit 62127206 (2022-04-26): Redirrect stderr from compiler into the tcl console (close #302) (#308)
closed 2022-04-26 by alain-rs