os-fpga/FOEDAG #330: License Requirements

coolbreeze413, 2022-05-01:
The License currently used by FOEDAG is GPL, which means that any derived work will be GPL.
From a legal standpoint, GPL considers any derived code, as well as any combination of components where one of the components is GPL (i.e. FOEDAG) then the combination, resulting in a product is also GPL.
Also, if there are 'data' files (device XML files for example), I believe that this would be outside the scope, as it is technically not a part of the product, and hence can be kept closed source.

However, it would be helpful to understand the implications of using this as part of a bigger combined product as to what can be kept closed w.r.t the GPL stipulations, hence raising this as an issue here.
Please advise.

alain-rs, 2022-05-03:
We put the GPL License initially because of QT. We should consult with QT to see what can be done differently.

coolbreeze413, 2022-05-03:
I had the same understanding, thanks for the cofirmation. 
Maybe LGPL would be more suitable, so that we can still leverage Qt as libs, and have a possibility for the rest of the code to have proprietary components, as we won't be modifying Qt itself.

alain-rs, 2022-10-19:
We need GPL because of Scintillia.

closed 2022-10-19 by alain-rs