os-fpga/FOEDAG #364: Project Settings, TCL Commands

coolbreeze413, 2022-05-12:
Do we have a plan for specific parameters in the project settings JSON file?
Some variables can already be set by TCL commands, if these are also defined in the JSON file, would it be appropriate for the TCL command to override the JSON file settings?

[counter_16bit.json.txt](https://github.com/os-fpga/FOEDAG/files/8682167/counter_16bit.json.txt)
A example preliminary JSON file that we are currently using is attached to provide more clarity on a possible way to organize the parameters.
`VPR` parameters are separate as of now, and they should go into the appropriate `Task` Settings - it is not yet clear on how to neatly categorize them.

alain-rs, 2022-05-14:
@coolbreeze413, please talk to @skyler-rs about this issue skyler@rapidsilicon.com
I tried to send you an email but is bounced.

coolbreeze413, 2022-05-17:
@skyler-rs : Thanks for the details!
Using the "id" as a key instead of a value in the "id" key makes sense and is good.
I understood the changes and will make them shortly in the adapted JSON.

As for the TCL commands - it would make sense for these to 'override' the values set from the settings JSON if any, with the implementation details according to when it fires (it depends on when the settings json object is created, for current FOEDAG, this would be: first json creation, and then TCL commands I think).

@alain-rs : To narrow down the particular idea between settings JSON and TCL: we may want to set the the 'name', 'foundry' etc in the settings JSON as the 'default/fallback', but can specify them via TCL commands as well - so that we can run the same designs for multiple families with a 'default/fallback' settings.json file.
What would be the direction FOEDAG will take on these kind of parameters ("device name", "foundry" etc. which identify a target configuration among multiple to run the analysis/bitstream generation) ?

coolbreeze413, 2022-05-17:
@skyler-rs : For the GUI creation, I had an idea which may make the whole Settings GUI a bit more generic:

If the only rule were to be, for every parameter-value in the settings.json to belong to a 'category' and 'subcategory', then, we can have a single dialog settings GUI, which will just loop through the 'category' + 'subcategory' and create appropriate widgets (and groups) according to the parameter type as it is currently.
Collapsible widget groups (according to category/subcategory) + current widget creation,  should make the UI good to use.

This way, any param-value in the settings.json can be represented in the GUI, instead of only those inside 'Tasks' as currently.
I have a basic implementation of this idea, and I will publish it, probably tomorrow, so you can have a look at it, to see if this approach has some merit to be used here.
I just wanted to run the idea by you to get your feedback.

skyler-rs, 2022-05-17:
@coolbreeze413 I was considering something similar. The widget generation is written in layers so any 1 widget can be generated w/ json parameters, a collection of json widget parameters can be created within a widget that contains a QDialogButtonBox w/ accept/cancel that checks if values have changed, and then that widget can be added to a dialog for the current implementation with tasks.

Where these settings should be stored (under Tasks, or somewhere more generic) is still open to exploration and the tasks implementation is just an initial attempt to see how things work out.

Also, please note that I have not been able to check in my refactor yet as I ran into some test/integration failures that I'm still trying to resolve. Once those changes are in, the EditTaskSettings command should demonstrate my current idea for the tasks specific implementation, but it could be pointed at different sections in the json pretty easily.

coolbreeze413, 2022-05-17:
@skyler-rs Thanks for the update, will wait for the refactor before making the changes.

skyler-rs, 2022-05-23:
@coolbreeze413 Changes are finally in with https://github.com/os-fpga/FOEDAG/pull/392.

I'm now exploring where to find the options for each task and where in the flow they should be applied.

skyler-rs, 2022-07-26:
Closing this as I think the initial conversation has been resolved w/ the initial implementation #488. Additional features or changes to the system cam be tracked through a new issue.

closed 2022-07-26 by skyler-rs