os-fpga/FOEDAG #381: Cannot edit correctly a Tcl file that has errors

alain-rs, 2022-05-16:
Create a project file (Tcl) that has a typo and load it in the gui. Click on the error hyperlink in the console.
The file will be marked with an error icon on the line of the error.
Now try to correct the error using the builtin editor:
Use the mouse to select the character to correct, most likely you will see another character gets edited when typing. Some weird things happen with character insertion. 
This does not happen on a regular file (When there is no error annotation).

KochynVolodymyr, 2022-05-17:
Can't reproduce. What do you mean 'load it in the gui' ? run with --script or use source command?

alain-rs, 2022-05-17:
broken.tcl:
create_design_bad DESIGN

./build/bin/foedag --script broken.tcl 

click on the Tcl stack in the console , go to the foedag text editor, try to correct the text create_design_bad into create_design  
you'll see.

ravic-rs, 2022-09-09:
@alain-rs, @zaryabkhan-rs couldn't reproduce this again recently. If it's still an issue please let us know.

His testing notes:
_create bgm.tcl file with text “create_design_BAD DESIGN”
Run ./build/bin/foedag --script bgm.tcl
It opened the Gui with error on the console with hyperlink on “bgm.tcl”.
Opened the hyperlink that opens the tcl file in the gui editor.
changed the text from  “create_design_BAD DESIGN” to “create_design design” and saved the file from editor.
Run again ./build/bin/foedag --script bgm.tcl and it worked as it is expected i.e it creates a design with name “design._

alaindargelas, 2022-09-10:
Close It

closed 2022-09-10 by ravic-rs