os-fpga/FOEDAG #431: Build break

alain-rs, 2022-06-01:
collect2: error: ld returned 1 exit status
gmake[3]: *** [CMakeFiles/foedag-bin.dir/build.make:223: bin/foedag] Error 1
gmake[3]: Leaving directory '/home/alain/FOEDAG/dbuild'
gmake[2]: *** [CMakeFiles/Makefile2:593: CMakeFiles/foedag-bin.dir/all] Error 2
[100%] Linking CXX executable ../../bin/console_test
/usr/bin/ld: ../../lib/libfoedagcore.a(Tasks.cpp.o):/home/alain/FOEDAG/src/Main/Tasks.cpp:37: multiple definition of `DBG_PRINT'; ../../lib/libfoedagcore.a(Settings.cpp.o):/home/alain/FOEDAG/src/Main/Settings.cpp:32: first defined here
/usr/bin/ld: ../../lib/libfoedagcore.a(WidgetFactory.cpp.o):/home/alain/FOEDAG/src/Main/WidgetFactory.cpp:37: multiple definition of `DBG_PRINT'; ../../lib/libfoedagcore.a(Settings.cpp.o):/home/alain/FOEDAG/src/Main/Settings.cpp:32: first defined here
collect2: error: ld returned 1 exit status
gmake[3]: *** [src/Console/CMakeFiles/console_test.dir/build.make:165: bin/console_test] Error 1
gmake[3]: Leaving directory '/home/alain/FOEDAG/dbuild'
gmake[2]: *** [CMakeFiles/Makefile2:1897: src/Console/CMakeFiles/console_test.dir/all] Error 2
gmake[2]: Leaving directory '/home/alain/FOEDAG/dbuild'
gmake[1]: *** [Makefile:160: all] Error 2
gmake[1]: Leaving directory '/home/alain/FOEDAG/dbuild'
make: *** [Makefile:53: debug] Error 2

alain-rs, 2022-06-01:
I merged 2 PR and got this after make clean.
Not sure why the CI does not catch it.

skyler-rs, 2022-06-01:
I think i can fix it by renaming my DBG_PRINT function to something less generic. I have to do an interview at 7, but i'll try to fix after that.

skyler-rs, 2022-06-01:
Change is in with https://github.com/os-fpga/FOEDAG/pull/432 altho i can't confirm if it worked as i wasn't able to recreate the build error. My main branch wasn't showing my changes even after a git pull so i had to using my existing branch from the previous PR. Apologies if this complicates anything.

alain-rs, 2022-06-01:
Fixed by #432

closed 2022-06-01 by alain-rs