os-fpga/FOEDAG #448: Integrate Synthesize Options with Settings Dialog

skyler-rs, 2022-06-06:
- [ ] The settings dialog should be able to read and display pre-set options that have been set via tcl commands/scripts and stored in the compiler object (SynthOpt() and SynthMoreOpt()).
- [ ] Values set by the settings dialog (and therefore stored in settings) should be passed to the related task when it is executed
- [ ] The following settings should be captured in the settings.json file for the Synthesis Task:

    _Settings captured from: raptor --help_
```
  synthesize <optimization>  ?clean? : RTL Synthesis, optional opt. (area, delay, mixed, none)
   synth_options <option list>: RS-Yosys Plugin Options. The following defaults exist:
                              :   -effort high
                              :   -fsm_encoding binary if optimization == area else onehot
                              :   -carry no_const
     -effort <level>          : Optimization effort level (high, medium, low)
     -fsm_encoding <encoding> : FSM encoding:
       binary                 : Compact encoding - using minimum of registers to cover the N states
       onehot                 : One hot encoding - using N registers for N states
     -carry <mode>            : Carry logic inference mode:
       all                    : Infer as much as possible
       no_const               : Infer carries only with non constant inputs
       none                   : Do not infer carries
     -no_dsp                  : Do not use DSP blocks to implement multipliers and associated logic
     -no_bram                 : Do not use Block RAM to implement memory components
     -fast                    : Perform the fastest synthesis. Don't expect good QoR.
```

skyler-rs, 2022-06-06:
Hey @alain-rs , a couple questions came up while planning this:

- Should there be default selections for these dropdown/1ofN options or should the user be able to leave an option unset and the value isn't passed to the system? Ex: does fsm_encoding require a value or can it be omitted sometimes. If the value can be omitted should the system provide a none/blank gui option to imply the parameter isn't passed?
- Do we want to represent the values exactly as the command line shows, or format them for UI. Example should onehot be "One Hot" in the dropdown menu in the ui or should it stay "onehot" to reflect the command line usage?

alaindargelas, 2022-06-06:
@skyler-rs, No default, default is no selection. You can beautify the name in the UI form.

alain-rs, 2022-06-10:
@skyler-rs, any update on this feature?

skyler-rs, 2022-06-10:
@alain-rs currently i've updated settings dailogs to read from the synth tcl options and use those if they are available or otherwise use system/user values if ones have been saved. I noticed an issue where create_design could change the projectPath without reloadSettings getting called so I added some new logic to try to catch all path changes. Synth options are all captured and I was testing to make sure the different layers of settings (tcl/defaults/user) were working and then i had to shelve my work to investigate the windows build issues.

So once the windows stuff is resolved, I need to finalize my logic to transfer the settings values to the synthesis command. I wrote some prototype stuff the other week, but wanted to confirm with you that my method made sense. I had written it before this issue was created so the logic probably needs some slight updates to match the requirements etc.

closed 2022-06-16 by alain-rs