os-fpga/FOEDAG #464: New Project menu pick is disabled after finishing project creation

skyler-rs, 2022-06-16:
> **Describe the bug**
> After finishing the new project wizard, the "New Project" menu pick is disabled until app restart.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Click File > New Project
> 2. Click through the creation steps and click finish
> 3. Click File
> 4. Note that "New Project" is no longer selectable
[image]

> **Expected behavior**
> The menu pick shouldn't be disabled and the user should be able to re-run the New Project wizard, creating and loading the new project. (Note, just from observing it's state when opening/closing it mid use, it looks like this might be one dialog instance so re-running the project wizard might take some refactoring.

skyler-rs, 2022-06-16:
Initial gui task for Zaryab once he starts working in the code

skyler-rs, 2022-06-16:
Note: This might be a dupe of https://github.com/os-fpga/FOEDAG/issues/218

KochynVolodymyr, 2022-06-16:
This is actually done. After project being closed - new project become available.

KochynVolodymyr, 2022-06-16:
I guess so

skyler-rs, 2022-06-17:
Gotcha, i'll close this for now as I'm not sure the additional logic would really add much usability-wise and probably isn't a good starting task for Zaryab.

closed 2022-06-17 by skyler-rs