os-fpga/FOEDAG #471: Usability Improvement: Allow Resetting Raptor Workflow Steps Incrementally

tmcdowell-rs, 2022-06-17:
This is a cross-post from the private Rapid Silicon GitHub, issue #93 

**Is your feature request related to a problem? Please describe.**
Allow a user to incrementally "roll back" the phases of the implementation workflow

**Describe the solution you'd like**
Allow users to reset the workflow back to some intermediate step. For example, if they have run all the way through bitstream generation, allow them to reset back to post-synthesis such that synthesis results remain coherent but place, route, timing analysis, and bitstream generation are re-run.

Similarly, allow the user to completely clean out the entire Raptor project similar to make clean.

**Describe alternatives you've considered**
I have tried to manually interact with the component tools (YoSYS, VPR, etc) but this does not update the FOEDAG state.

**Additional context**
Competitively, Xilinx Vivado has commands for both of these operations.
To reset a run back to a particular steps the reset_runs Tcl command is available in Vivado
To reset the entire project workspace the reset_project Tcl command is available in Vivado

tmcdowell-rs, 2022-06-17:
Please assign to @KochynVolodymyr per request from @alain-rs. I don't have appropriate permissions on this repo.

alain-rs, 2022-06-17:
@tmcdowell-rs, you are added to the repo

KochynVolodymyr, 2022-06-17:
I've already added clean commands for compiler steps, you can find them in the Task table. Is that what you need?

tmcdowell-rs, 2022-06-17:
@KochynVolodymyr - per the other item in the RS repo, there was a bug in the behavior that might be what @alain-rs was intending to be solved in upstream.  For simplicity, I've reproduced it here:

----------
This looks like there's still something missing.  If I clean synth (for example), the downstream items are still showing as valid.  The downstream results of any step should also be invalidated from my experience in Vivado (unless something is different in our flow).  Should I re-open this bug, open a new bug, or mark this as a documentation thing?
[image]

KochynVolodymyr, 2022-06-17:
This is old, try the latest

KochynVolodymyr, 2022-07-11:
@tmcdowell-rs please give feedback here.

commit c22016c9 (2022-07-13): Invalidate downstream compiler tasks when user press clean (close #471) (#512)