os-fpga/FOEDAG #473: Since file path separator "/" uses everywhere it might not work on WIN.

KochynVolodymyr, 2022-06-20:
If search through the project "/" there are many entries. They are related to some paths. It might not work for the Window OS.

alain-rs, 2022-06-21:
Use std::filesystem::path apath =   "a" / "b" / "c" ;   the / operator will create the right separator automatically.

zaryabkhan-rs, 2022-10-07:
IP configurator is not working properly in Windows build. 
**Screenshot windows build**
[image]

- It doesn't show the IPs.
- When Generate IP is pressed then, No folder/files are created in project Directory.

ravic-rs, 2022-10-07:
> @ravic-rs IP configurator is not working properly in Windows build. **Screenshot windows build** [image]
> 
> * It doesn't show the IPs.
> * When Generate IP is pressed then No folder/files are created in project Directory.

We don't have any IPs available other than the Mock ones, I will let @skyler-rs comment on it.

zaryabkhan-rs, 2022-10-07:
> We don't have any IPs available other than the Mock ones, I will let @skyler-rs comment on it.

Yeah but we have two Mock Ips shows in ubuntu build. It also suppose to be shown in windows.

ravic-rs, 2022-10-07:
We won't be able to tell if this breaks until Windows build is supported.

skyler-rs, 2022-10-07:
@zaryabkhan-rs are there any errors printed in the tcl console? In particular, in foedag, which doesn't have a python venv installed, you should be seeing an error telling you a local python interpreter couldn't be found and then it printing what it found using LocateExecFile, but [LocateExecFile fails on windows currently](https://github.com/os-fpga/FOEDAG/issues/661) so my initial guess is this is failing at the finding python step, but can't be sure from that image.

It does look like the output Dir has a bad path display; however, i think i hardened the actual filepaths at the script execution point, but I can definitely look into fixing it earlier at this point as well.

I haven't tried a windows build recently, but as @ravic-rs mentions, eventually i'll probably need to build and debug this in windows so that could be a gating factor.

alain-rs, 2022-10-07:
Use the filesystem class separator operator that works on all OSs. There are plenty of working example code in the Compiler subsystem.

[image]<https://user-images.githubusercontent.com/109368108/194475171-cab54c76-8402-47d4-aae8-eb9d9a7335b0.png>