os-fpga/FOEDAG #522: Status column width mismatch under Task tab

ravic-rs, 2022-07-14:
> **Describe the bug**
> The status column width within the "Task" tab is not set correctly.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone repository and checkout commit id: 330f3800b48c (on main branch).
> 2. Build and launch foedag.
> 3. View the Status column under task

> **Expected behavior**
> Status column width should atleast show the entire text "Status" clearly to user. It should behave well when windows are resized.

> **Screenshots**
> [image]

> **Environment (please complete the following information):**
sw03 on server but likely present in all environments.
>  - OS: 
[NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"
CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"]

>  - Compiler:
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)

> - Version:
 330f3800b48c (on main branch)

ravic-rs, 2022-07-19:
Please check on the latest commit and verify on upstream GUI as I see it there too.

TarasSydor, 2022-07-20:
Ravi, I can't reproduce it anymore. Regular startup and with --script command works OK and Status width is correct.
Were there some UI manipulations you did to reproduce it? Tasks view doesn't allow to shrink its width so that columns don't fit.

ravic-rs, 2022-07-20:
> Ravi, I can't reproduce it anymore. Regular startup and with --script command works OK and Status width is correct. Were there some UI manipulations you did to reproduce it? Tasks view doesn't allow to shrink its width so that columns don't fit.

I can see this both foedag and upstream GUI on the **server**. Please check the build there.

closed 2022-08-02 by ravic-rs