os-fpga/FOEDAG #555: While manually resizing Task column in task view, button doesn't keep up

TarasSydor, 2022-08-01:
> **Describe the bug**
By default, all the columns are sized according to their content.
Task column allows manual resizing. While holding the mouse, text gets truncated according to the current width, but the
button remains stable till the user releases the button:

[image]

Status column doesn't have this problem because it has fixed width. Stats can also be resized, but doesn't have such button.
So it's just Task column.

TarasSydor, 2022-08-01:
The main question here: should we allow to shrink the Task columns so that text doesn't fit?
The issue is present regardless of the answer, as enlarging the column has the same effect - button remains stable till releasing the mouse. 
So unless we disable Task column resizing at all, the layout problem has to be fixed.

ravic-rs, 2022-08-26:
@TarasSydor you had made some progress on this issue a week or so ago. Is it ready for a PR ?

ravic-rs, 2022-09-01:
@KochynVolodymyr does your PR https://github.com/os-fpga/FOEDAG/pull/604/files affect this one ?

KochynVolodymyr, 2022-09-01:
> @KochynVolodymyr does your PR https://github.com/os-fpga/FOEDAG/pull/604/files affect this one ?

No, it isn't

ravikiranchollangi, 2024-03-08:
Closing this as we updated the widget now.

closed 2024-03-08 by ravikiranchollangi