os-fpga/FOEDAG #557: Save and load project path leads to incorrect work after project was moved.

KochynVolodymyr, 2022-08-02:
**Describe the bug**
Project path is saved to the file. This means that if project was moved - the path become invalid. E.g. Property window shows incorrect path to the files.

**To Reproduce**
1. Create some project.
2. Move project folder to another location.
3. Open project.

**Expected behavior**
Project path point to actual path.

ravic-rs, 2022-09-01:
We decided to remove the path from project file as it doesn't serve any purpose.

zaryabkhan-rs, 2022-09-02:
If we remove the path from project file and load that file to gui, then It didn't add the name of the design. 

[image]

KochynVolodymyr, 2022-09-02:
Because you need set project path from actual project file path but not read from file. If you are not doing that - we have a lot of problems.