os-fpga/FOEDAG #575: Foedag implementation of Compiler is missing clean logic

skyler-rs, 2022-08-12:
> **Describe the bug**
> The dummy version of Compiler in Foedag has fallen out of sync with the CompilerOpenFPGA copy and currently can be put into an infinite loop

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Launch foedag (this is not reproduceable in Raptor as it has the missing clean logic)
> 2. `create_design blah`
> 3. `place clean`
> 4. `place`

> **Expected behavior**
> `place clean` and `place` should fire once without issue (this is reproduceable on all tasks w/ a clean option besides `synthesize` whose foedag implementation already has clean logic)
> - This can be achieved by moving CompilerOpenFPGA clean logic into Foedag's copy of Compiler

> **Enviornment (please complete the following information):**
>  - OS: Ubuntu 20.04
> - Version: Main (FOEDAG 0.0.17)

KochynVolodymyr, 2022-08-12:
Are you sure this is necessary? A lot of things is missing in Compiler class.

skyler-rs, 2022-08-12:
It's causing an infinite loop without it.

closed 2022-08-26 by ravic-rs