os-fpga/FOEDAG #584: Integrate VPR GUI as a widget?

tangxifan, 2022-08-18:
**Is your feature request related to a problem? Please describe.**
VPR team is developing new features for their GUI (Visualizer for FPGA architecture)

Here are the thoughts proposed by Prof. Vaughn Betz and the student who is working on this project.

```
Vpr UI: use glade (graphical GUI builder to create all widgets) or use glade only for top-level organization (grid/layout) and code to create menus
Let me know if you have an opinion; can do either but Sebastian trying to figure out what is easiest for others for the future
```

**Describe the solution you'd like**
If we do want to think about integrating the GUI as a widget into FOEDAG, we should talk to them and see how the choice on GUI builder will impact the FOEDAG integration.

alain-rs, 2022-08-19:
@tangxifan can we try to convince them to switch to QT? If yes, we could integrate into Foedag as long as the VPR code is built as a component (library).
Are they aware of the Foedag framework?

alain-rs, 2022-08-22:
@ravic-rs, please get in touch with that team.