os-fpga/FOEDAG #590: RTL Library support

alain-rs, 2022-08-23:
http://yosyshq.net/yosys/cmd_verific.html supports Compilation Libraries. 
It is specially important for VHDL. Please read the Yosys documentation to understand the principles.
Add additional **options** to the Tcl command add_design_file:  -work  and   -L 
If -work is not specified, no need to pass the option to the Verific/Other parser, by default Verific/Other will compile under the "work" library (Equivalent to -work work).

Typical Foedag usage:
...
add_design_file -work lib1  file1.vhd file2.vhd
add_design_file -work lib2  file21.vhd file22.vhd
add_design_file -L lib1 -L lib2 top.vhd
...
Will translate to the Yosys script:
...
verific  -work lib1  -vhdl file1.vhd file2.vhd
verific  -work lib2   -vhdl  file21.vhd file22.vhd
verific  -L lib1 -L lib2 -vhdl  top.vhd
...

The GUI New Project should also support the Library concept. 
It is both usable for VHDL and Verilog.

KochynVolodymyr, 2022-08-26:
> add_design_file -L lib1 -L lib2 top.vhd
Do we expect 2 libraries sharp for '-L' parameter?
Should we support following command?
add_design_file -L lib1 -L lib2 -L lib3 top.vhd

Any number of libraries

TarasSydor, 2022-08-29:
We have Verific and default Yosys parser. Yosys one seem to work not in the way we expect: it gathers all the files added by all add_design_file() commands and puts them into a single one.
For example following tcl script:
add_design_file RotWord.sv  -SV_2012
add_design_file aes_decrypt128.sv
add_design_file generic_muxfx.v

results in following ys:
read_verilog  -sv  RotWord.sv aes_decrypt128.sv generic_muxfx.v 

Thus, in order to apply library arguments for each add_design_file() files, this scheme will have to be changed.
Verific parser, on the other hand, seem to behave differently and keeps files from various commands separated.

KochynVolodymyr, 2022-08-29:
@alain-rs  do we need to apply this to read_verilog as well? If not, what should happened if we don't use verific?

alaindargelas, 2022-08-29:
Yes, read_verilog needs also to receive the files that got grouped together into a compilation unit.

add_design_file RotWord.sv -SV_2012
add_design_file aes_decrypt128.sv -SV_2012
add_design_file generic_muxfx.v

becomes:
read_verilog -sv RotWord.sv 
read_verilog -sv aes_decrypt128.sv
read_verilog generic_muxfx.v

On the other hand, read_verilog does not support the -work and -L concepts. We should warn the user that those arguments are ignored when using Yosys default parser.

KochynVolodymyr, 2022-09-01:
import also required -L parameters: 
verific -L lib1 -import ...

All -L libraries need to pass to the import command as well.

TarasSydor, 2022-09-01:
Putting all -L libraries to import command still doesn't work for me. Input:
add_design_file -work lib1 aes_decrypt128.sv  -SV_2012
add_design_file -L lib1 generic_muxfx.v wrapper.v
Output:
verific -work lib1  -sv2012 /home/work/Work/Raptor/tests/Testcases/aes_decrypt_fpga/aes_decrypt128.sv
verific -L lib1  -vlog2k /home/work/Work/Raptor/tests/Testcases/aes_decrypt_fpga/generic_muxfx.v /home/work/Work/Raptor/tests/Testcases/aes_decrypt_fpga/wrapper.v
verific -L lib1 -import wrapper

and the following error goes off: wrapper.v:34: instantiating unknown module 'decrypt'

Changing '-work' in the first add_design_files to '-L' does compiler correctly, but I'm not sure whether it makes sense.
Should 'work' libraries also end up in import command?

ravic-rs, 2022-10-05:
@TarasSydor please link your PRs associated with this issue so we can close it.

TarasSydor, 2022-10-05:
Done.
I couldn't link the one below, so adding explicit link to it:
https://github.com/os-fpga/FOEDAG/pull/603

closed 2022-10-17 by alain-rs