os-fpga/FOEDAG #593: Fix MSVC Compiler version in Cmake to reflect Github Action update

alain-rs, 2022-08-24:
@ravic-rs, urgent.
Don't merge anything until this is fixed.

alain-rs, 2022-08-24:
That is why Tcl build is failing.

ravic-rs, 2022-08-24:
> That is why Tcl build is failing.

@alain-rs where are we setting the msvc version in cmake ? We didn't update foedag github actions recently, is it coming from github side?

https://github.com/os-fpga/FOEDAG/blob/main/.github/workflows/main.yml#L364
https://github.com/os-fpga/FOEDAG/blob/main/CMakeLists.txt
https://github.com/os-fpga/FOEDAG/blob/main/third_party/tcl8.6.12/.github/workflows/win-build.yml

alain-rs, 2022-08-24:
@ravic-rs, Github update the compiler every once in a while.
We have to follow suit, else it break. Last time this happened:

https://github.com/os-fpga/FOEDAG/pull/372/files

Do a grep for the compiler version tag to make sure there is a single place we need to update.
If you find a more robust way (Interrogate the compiler version to build the path) please implement it.

ravic-rs, 2022-08-24:
> @ravic-rs, Github update the compiler every once in a while. We have to follow suit, else it breaks. Last time this happened:
> 
> https://github.com/os-fpga/FOEDAG/pull/372/files
> 
> Do a grep for the compiler version tag to make sure there is a single place we need to update. If you find a more robust way (Interrogate the compiler version to build the path) please implement it.

Thanks for the details. I will take a look.

closed 2022-08-25 by ravic-rs