os-fpga/FOEDAG #597: Add Windows build documentation

ravic-rs, 2022-08-25:
**Is your feature request related to a problem? Please describe.**
Documentation update due to https://github.com/os-fpga/FOEDAG/pull/589

**Describe the solution you'd like**
Update doc files as needed to help user to install foedag under Windows OS.

**Describe alternatives you've considered**
NA

**Additional context**
Add any other context or screenshots about the feature request here.

coolbreeze413, 2022-08-25:
@ravic-rs : I have added a PR with the doc update.
This updates the existing doc with the dependencies setup using the MSYS2 env.
As build/install steps would remain the same as default (`make` or `make release` etc.) I have not modified any other documents.
Please let me know if it looks ok.

ravic-rs, 2022-08-26:
> @ravic-rs : I have added a PR with the doc update. This updates the existing doc with the dependencies setup using the MSYS2 env. As build/install steps would remain the same as default (`make` or `make release` etc.) I have not modified any other documents. Please let me know if it looks ok.

Thank you, looks good. We will try those steps later and get back if more details are needed.

closed 2022-08-26 by ravic-rs