os-fpga/FOEDAG #600: Add "analyze" Tcl command

alaindargelas, 2022-08-26:
This optional command and Compilation task needs to be added.
It is expected to run this way in a typical script:

```
# Design
create_design and2_gemini
add_design_file -V_2001 ./rtl/and2.v
add_constraint_file constraints.sdc

# Device target
target_device GEMINI

analyze     << This command will dump on the screen some information like: list of top level modules (In the case there is more than one).

set_top_module and2

# Compilation
synthesize delay
packing
place
route
sta
bitstream 
```

The analyze command will also return the list of top level modules as it's Tcl return value.

This command will run a design analysis step (Verific or Other parser based).
Several Json data dumps will be generated during that run: 
   - Top level module list
   - Port information for all top-level modules
   - Hierarchy tree for all top-level modules

The GUI will present that information in different views after either the tcl command is invoked or the "Analysis" Task is clicked.

alain-rs, 2022-08-26:
@aram-rs, this is the command that will encapsulate the new analyse executable. Please call the axecutable "analyze" for consistency.

alain-rs, 2022-09-09:
@volodymyrkochyn @ravic-rs 
Now that we have the analyze executable ready and integrated, we need to parse in the GUI the .json file.
There are several usages for the data:
 - one is for the pin planner, 
 - another use are new Tcl commands (Yet to be added) to interrogate that data from Tcl:

get_top_modules : returns in a Tcl list the list of top-level modules found by the analyze command
get_top_ports   : returns in a Tcl list a list of top-level RTL ports, size, direction and data type
get_hierarchy -top <top> -level <n> : returns a Tcl list of lists the hierarchy from a given top level up to a certain number of levels. 

Please parse the .json data once after the analyze executable is ran and save the data into a C++ data-structure that will allow to write such Tcl commands and use the data in the Pin Planner.