os-fpga/FOEDAG #632: Windows build failed when run make release command

zaryabkhan-rs, 2022-09-08:
> **Describe the bug**
Windows build failed after installing all dependencies and adding environment variables as instructed in the foedag documentation in dependencies section. It failed when running the make "release command " after all environment configuration 

> **To Reproduce**
> Steps to reproduce the behavior:
>-	Install Visual Studio Community Edition
o     From Installer, Check "Desktop development with C++"
o	Ensure "CMake tools for windows is checked"
>-	Install QT5 for MSVC
o	I tried  https://download.qt.io/official_releases/qt/5.12/5.12.12/qt-opensource-windows-x86-5.12.12.exe from  
        https://www.qt.io/offline-installers (there might be easier paths/better installers for this)
o	When asked to select components,  checked MSVC 2017 32 and 64 bit

>-	Update Environment Variables
o	System Properties > Advanced > Environment Variables…
o	Edit Path and add a new entry for Qt5, mine was C:\Qt\Qt5.12.12\5.12.12\msvc2017_64\bin
o	Create a new Variable called Qt5_DIR and point it to cmake/Qt5, mine was C:\Qt\Qt5.12.12\5.12.12\msvc2017_64\lib\cmake

>-	Install Make Package by following command for windows:
Steps for Installing "windows make":
o	Installed make using "winget install gnuwin32.make" on cmd
o	Select environment variable under the Advanced tab.
o	Under the variable name, enter make.
o	Under Variable as "make" and  value as , C:\Program Files(x86)\GnuWin32\bin\make.exe.

>- MSYS2 env with the MinGW64 g++ compiler can be used to build FOEDAG as mentioned in the documentation.
>- git clone https://github.com/os-fpga/FOEDAG.git
>- cd FOEDAG
>- git submodule update --init --recursive

>- make release
>-ERRORS: 
> **Screenshots**
[image]

> **Expected behavior**
> FOEDAG build successfully in Windows environment 

> **Additional context**
> Add any other context about the problem here.

ravic-rs, 2022-09-08:
@coolbreeze413 can you please take a look at what we are missing ?
https://foedag.readthedocs.io/en/latest/tutorials/getting_started/dependencies/#win

coolbreeze413, 2022-09-08:
I have not tried the msvc build for some time, we are only using the msys2 env.

I remember that for msvc, we need to initialize env on command line using vcvarsall.bat, before cmake invocation, for 2019 this was the path I used:
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat

coolbreeze413, 2022-09-08:
@ravic-rs I can see this step in the workflow: 
https://github.com/os-fpga/FOEDAG/blob/189a92304646502254eb425939038a48f7f4266a/.github/workflows/main.yml#L316

I think you would need to execute that before cmake.

ravic-rs, 2022-09-08:
@zaryabkhan-rs please work with @coolbreeze413 on resolving this one.

zaryabkhan-rs, 2022-09-12:
Hey @coolbreeze413, Can you share  instructions related to msys2 env with Windows version(10 or 11) you used?

zaryabkhan-rs, 2022-09-12:
> @ravic-rs I can see this step in the workflow:
> 
> https://github.com/os-fpga/FOEDAG/blob/189a92304646502254eb425939038a48f7f4266a/.github/workflows/main.yml#L316
> 
> I think you would need to execute that before cmake.

I used the same before cmake. but got those errors. I have checked on windows 11 and windows 10.

coolbreeze413, 2022-09-12:
@zaryabkhan-rs the steps to setup msys2 are here: https://github.com/os-fpga/FOEDAG/blob/main/docs/source/tutorials/getting_started/dependencies.rst#msys2-mingw64

BTW, were you able to compile using msvc with the addition of the vcvars script execution?

zaryabkhan-rs, 2022-09-12:
> @zaryabkhan-rs the steps to setup msys2 are here: https://github.com/os-fpga/FOEDAG/blob/main/docs/source/tutorials/getting_started/dependencies.rst#msys2-mingw64
> 
> BTW, were you able to compile using msvc with the addition of the vcvars script execution?

No, It tried that but it didn't work on windows 11. I also checked on win 10 but result was same. Does msys2 env work on window 10 only?

zaryabkhan-rs, 2022-09-13:
> > @zaryabkhan-rs the steps to setup msys2 are here: https://github.com/os-fpga/FOEDAG/blob/main/docs/source/tutorials/getting_started/dependencies.rst#msys2-mingw64
> > BTW, were you able to compile using msvc with the addition of the vcvars script execution?
> 
> @coolbreeze413  No, It tried that but it didn't work on windows 11. I also checked on win 10 but result was same. Does msys2 env work on window 10 only?

coolbreeze413, 2022-09-13:
@zaryabkhan-rs msys2 works on both windows 10 and 11, same steps.

zaryabkhan-rs, 2022-09-13:
@coolbreeze413  I have installed all required installation for msys2 mentioned in the link as shared above, Now to build, Do we just need to run make release? or Is there any other thing required related to env variable or vcvars script execution etc.?

zaryabkhan-rs, 2022-09-13:
@coolbreeze413  Can you verify what we are missing? I have followed these steps:
1) Everything mentioned here  https://github.com/os-fpga/FOEDAG/blob/main/docs/source/tutorials/getting_started/dependencies.rst#msys2-mingw64
2) make release

I got this output
[image]

coolbreeze413, 2022-09-14:
@zaryabkhan-rs It looks like the system where you are building has some environment variables hardcoded (probably set in the Windows environment variables).

For MSYS2 environment, we need to use the `MSYS2 Makefiles` generator and the `gcc/g++` compiler instead of `ninja` and `MSVC` as it seems to be set currently.

Ideally, this should be the default, but most likely the env on your system has changes which are changing these defaults.

I suggest to try first:

- set the cmake generator: `export CMAKE_GENERATOR=MSYS Makefiles`

and then try to do the `make release`
This should automatically select gcc/g++ as the compiler as well.

However, if you still see the compiler being selected as cl and erroring out, then you can additionally try to set the compiler:

- `export CC=gcc`
- `export CXX=g++`

And then build using `make release`

zaryabkhan-rs, 2022-10-06:
**Summary** 
Following steps for MSYS32 env build works for me.
OS : Windows 11

**MSYS2 MinGW64**
MSYS2 env with the MinGW64 g++ compiler can be used to build FOEDAG.

Get the latest installer from : [MSYS2](https://www.msys2.org/) 

Follow the steps listed below, step 1-6), and step 7 lists the additional packages needed to build FOEDAG:

Invoke the downloaded installer

Allow installer to run the MSYS2 MinGW64 Shell

Run pacman -Syu for initial base updates

At the end, it will close the terminal after confirmation

Run MSYS2 MinGW64 Shell

Run pacman -Su for remaining base updates

Run pacman -S --needed base-devel mingw-w64-x86_64-toolchain git mingw-w64-x86_64-cmake mingw-w64-x86_64-qt5-base-debug mingw-w64-x86_64-qt5 mingw-w64-x86_64-swig mingw-w64-x86_64-qt5-declarative-debug mingw-w64-x86_64-tcl mingw-w64-x86_64-zlib for installing required packages.

Select default (all) packages to install here

Close the MSYS2 MinGW64 Shell

Now, Run again MSYS2 MinGW x64 Shell (from Start Menu) and start building.

run make.

**Note**: make sure for this env, it should point out to MinGW Qt libraries. if you face any linker problem while building, try to set these option and then run make. cmake must point to MinGW Qt libraries.

export Qt5_DIR=/mingw64/lib/cmake/Qt5

closed 2022-11-16 by ravic-rs