os-fpga/FOEDAG #640: Once project is created, it cannot be changed

alain-rs, 2022-09-09:
If you add the wrong constraint file while creating a project in the GUI, it cannot be removed. Worse, the widget that is supposed to show how to edit the constraint specification does not show the original constraint file
[image]

alain-rs, 2022-09-09:
@ravic-rs, please make sure that all the project settings can be edited, files, device target... once the project is created and re-opened for edits.

KochynVolodymyr, 2022-09-09:
This widget just show new files but not the old. This is how it designed. But of you want change order in your design - no chance. There is bug with remove.

KochynVolodymyr, 2022-09-09:
No chance to change target device

alaindargelas, 2022-09-09:
We need everything to be editable.

ravic-rs, 2022-09-09:
We will take a look.

KochynVolodymyr, 2022-09-14:
@alain-rs 1. User are able to remove file via context menu. 2. This window is only for adding or creating new files. Like in vivado.

alain-rs, 2022-09-19:
Closing this issue, wizard issues are tracked with an internal Jira.

closed 2022-09-19 by alain-rs