os-fpga/FOEDAG #1133: Detailed utilization report according to functional modes

tpagarani, 2023-05-31:
**Is your feature request related to a problem? Please describe.**
Current utilization reports do not give details about how each CLB is utilized. 

**Describe the solution you'd like**
each CLB can be used in either 6-LUT mode or 2-LUT5 mode or an ADDER mode. 
Also each FF can be used in LUT+FF mode or Direct FF. The report file should include how many CLBs are used in which mode
This information can be derived from VPR log file e.g. in following report
Pb types usage...

  clb                                   : 92
   fle                                  : 711
    ble6                                : 81
     lut6                               : 81
      lut                               : 81
     ff                                 : 6
      DFFSRE                            : 6
    lut5inter                           : 630
     ble5                               : 1090
      flut5                             : 1082
       lut5                             : 522
        lut                             : 522
       ff                               : 652
        DFFSRE                          : 652
      soft_adder                        : 8
       adder                            : 8
        lut4                            : 2
         lut                            : 2
        adder_carry                     : 8 

**Describe alternatives you've considered**
Currently we have to parse the VPR and Yosys log file to get these details

**Additional context**
Add any other context or screenshots about the feature request here.

alain-rs, 2023-08-23:
@ravic-rs FYI

ravikiranchollangi, 2024-02-03:
@coolbreeze413 are you guys ready to PR this ?