os-fpga/FOEDAG #1479: IPA-FOEDAG integration

ravikiranchollangi, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

w0lek, 2024-03-04:
hi @ravikiranchollangi i have some access problems to push my changes to github.
as soon as i will have access right i will push changes.

w0lek, 2024-03-04:
DONE:
1) the latest IPA code base  (which is using in Aurora v2.6 release) is migrated to FOEDAG.
2) The IPA widget is connected to FOEDAG and UI entry point is set

Current limitation:
1) we need to have modified vpr binary supporting --server mode and multiselection functionality
2) we need get proper CompilerFPGA::BaseVprCommand to be used for IPA. This is part of TODO_IPA_MIGRATION_COMPILER definition reviewing.

So, left to TODO:
1) solve TODO_IPA_MIGRATION_COMPILER definition to be able to run vpr session
2) solve TODO_IPA_MIGRATION_SETTINGS definition to be able to integrate IPA settings to project settings.
Note: probably without solving it it still should work but with saving to local json file, this is needs to be checked
3) integrate IPA unitests
4) provide modified vpr in order to run it in server mode and use FOEDAG IPA as a source control.
This will be done in openfpga fork branch/PR

todo points 1) and 4) are mandatory to have IPA functioning
todo points 2) and 3) are not mandatory to have IPA functioning, so it could be done with lower priority

w0lek, 2024-03-08:
@ravikiranchollangi @volodymyrkochyn 
i need additional time to understand how to run P&R View from FOEDAG.
Maybe you could help me and speed up the procedure.
Right now what i am doing:
- i create FOEDAG project called project_1
- i copied aurora2 binaries from https://github.com/QL-Proprietary/aurora2/releases/tag/v2.6.0-beta
to build directory of FOEDAG. Note: this is for quick way to having modified vpr, to check it along with FOEDAG+IPA.
- then i run ./foedag --compiler openfpga
- i run IPA tool, and enable [log to file] in order to capture vpr cmdline which is used by IPA

in my case it's
`/home/work/workspace/repos/FOEDAG/build/libexec/vpr  /home/work/foedag_projects/project_1/run_1/synth_1_1/synthesis/project_1_post_synth.v --sdc_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/packing/fabric_project_1_openfpga.sdc --clock_modeling ideal --route_chan_width 100 --allow_unrelated_clustering on --net_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/packing/project_1_post_synth.net --place_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/placement/project_1_post_synth.place --route_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/routing/project_1_post_synth.route --analysis --server --port 60555 --disp on
`
when i copy paste this cmd to terminal, and try to run it, i have vpr error

`work@lvilwks0007:~/foedag_projects/project_1$ /home/work/workspace/repos/FOEDAG/build/libexec/vpr  /home/work/foedag_projects/project_1/run_1/synth_1_1/synthesis/project_1_post_synth.v --sdc_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/packing/fabric_project_1_openfpga.sdc --clock_modeling ideal --route_chan_width 100 --allow_unrelated_clustering on --net_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/packing/project_1_post_synth.net --place_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/placement/project_1_post_synth.place --route_file /home/work/foedag_projects/project_1/run_1/synth_1_1/impl_1_1_1/routing/project_1_post_synth.route --analysis --server --port 60555 --disp on
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+254d38faa-dirty
Revision: v8.0.0-8515-g254d38faa-dirty
Compiled: 2024-03-08T01:30:44
Compiler: GNU 9.4.0 on Linux-5.11.0-38-generic x86_64
Build Info: Release IPO VTR_ASSERT_LEVEL=2

University of Toronto
verilogtorouting.org
vtr-users@googlegroups.com
This is free open source code under MIT license.

Missing required positional argument: circuit

usage: vpr architecture circuit [--pack] [--place] [--route] [--analysis] 
       [--server] [--disp {on, off}] [--save_graphics {on, off}] 
       [--graphics_commands GRAPHICS_COMMANDS] [-h] [--version] 
       [--show_resource_usage {on, off}] [--device DEVICE_NAME] [-j NUM_WORKERS] 
       [--timing_analysis {on, off}] [--disable_errors DISABLE_ERRORS] 
       [--suppress_warnings SUPPRESS_WARNINGS] 
       [--terminate_if_timing_fails {on, off}] 
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]
`
so i need verify cmd line arguments for vpr, it says some required positional arguments are missing. Maybe you may give a hint, with real working sample of OpenFPGACompiler::BaseVprCmd()

volodymyrkochyn, 2024-03-08:
@w0lek few questions:
* why it's under `build/libexec/vpr`? It should be `build/bin/vpr`
* Did you make sure vpr is the same version?

you can try raptor, vpr works if start this command from command line

w0lek, 2024-03-22:
@volodymyrkochyn 
"why it's under build/libexec/vpr? It should be build/bin/vpr"
it's the way Aurora package is build on.
in build/bin/vpr we put the shell script instead of real binary with name "vpr", and this shell script do initialization of environment and uses vpr executable from build/libexec/vpr.
But anyway this is obsolete now, since we skip this stage.

w0lek, 2024-03-22:
@volodymyrkochyn 
@ravic-rs 
my status up to now.
i updated FOEDAG PR header and add few screenshots.
I found some FOEDAG_rs CI fail on msvc, so i push new commit into FOEDAG submodule to fix that.
@volodymyrkochyn could you restart the FOEDAG_rs please? it seems it doesn't start automatically.
@ravikiranchollangi to test Raptor CI, we probably need to do merge of FOEDAG_rs first, and  thant update the Raptor repository to use new reference in main, because now, i am using FOEDAG_rs fork insude the Raptor, and this brings the conflict and CI is not starting.

**Left to do:**
- merge PR FOEDAG
- check FOEDAG_rs CI finish and if it's ok, we could merge PR
- update Raptor PR to reference to right FOEDAG_rs hash and push changes
- check Raptor CI finishing
- if Raptor CI finish ok, we need set default IPA=0 in Raptor Makefile, commit and push again.
From VOlodymyr's discussion i understand that we could left IPA ON by default for FOEDAG and FOEDAG_rs repositories.

ravikiranchollangi, 2024-03-28:
@w0lek please check the latest raptor merges, I believe we can close this now.

w0lek, 2024-04-01:
Raptor has merged IPA changes, issue is closed

closed 2024-04-01 by w0lek